S7NetPlus/s7netplus #421: Request: Flush/Reset TCP Buffers

droessler-robotron, 2021-12-10:
Thanks for providing this lib and code! 
We are using this lib for a while and reading data form a real Siemens S7 PLC on different load situations... 

After periodically reading some data from fixed addresses for 2..3 days in fixed interval of about 1..3 seconds there is a known issue: 

`S7.Net.PlcException:“Received 15 bytes: '32-03-00-00-00-07-00-02-00-01-00-00-05-01-FF', expected 19 bytes.”` like here https://github.com/S7NetPlus/s7netplus/issues/384 described. 

And this, once occured, will not restore back to a state as before. This exception will now happen on and on. I am not really sure if it fails with every loop, but as far as i see it stays until i force a new connection manually. 

**I'd like to handle this Issue well and wonder if there is a possibility to flush (clear) the stream buffers from the TCP Connection?? **

By the way ... will this exception only occures when there are bytes missing or also when there are more bytes than expected ? 
 
 
 have a nice weekend! regards Tobias

scamille, 2021-12-10:
Which version of S7NetPlus are you using?

droessler-robotron, 2021-12-14:
oh yes, sorry. It's version "0.8.1"

scamille, 2021-12-14:
The reason is that at some point I did something to properly reset the connection on those errors in #349 

Please try again with 0.13.0

droessler-robotron, 2022-01-11:
We upgaded to  0.13.0 and for the first there was no further error...   we observe it for long time stability. 

Thank you very much and I think we close this issue.

closed 2022-01-11 by droessler-robotron