S7NetPlus/s7netplus #424: Unexpected Values

bgreer5050, 2022-01-09:
I am using the sample code from the Wiki:

// reads a class from DataBlock 1, startAddress 0
TestClass myTestClass = new TestClass();
plc.ReadClass(myTestClass, 1, 0);

This is working well for some of my values.  Some values that are coming back do not make sense so I think I need to convert them to something else but I am not sure to what.  For example, I have some values that should always be positive but are sometimes negative.  I am guessing that either I am using the wrong type in my struct.  

I will show my datablock and then a copy of my struct.  Please point out where I might be using the wrong type in my struct.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/bgree/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/bgree/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Address | Name | German   Name | Type | Initial   Value | Actual Value
-- | -- | -- | -- | -- | --
0.0 | Req_GUD | Allgemein.Req.GUD | BOOL | FALSE | FALSE
0.1 | GUD_Error | Allgemein.GUD_Error | BOOL | FALSE | FALSE
0.2 | GUD_NDR | Allgemein.GUD_NDR | BOOL | FALSE | FALSE
2.0 | Variables_Counter_Absolute | Variables_stueckzaehler_absolut | DINT | L#0 | L#32150147
6.0 | Variables_Counter_Goal | Variables_stueckzaehler_soll | DINT | L#0 | L#105
10.0 | Variables_Counter_Is | Variables_stueckzaehler_soll_ist | DINT | L#0 | L#105
14.0 | Variables_Pipe_Length | Variables_ROHRLAENGE | REAL | 0.000000e+000 | 524.637
18.0 | Variables Measured Pipe Length | Variables_GEMESSENE_ROHRLAENGE | REAL | 0.000000e+000 | 517.6703
22.0 | Variables_SCB_RLM_MK1 | Variables_SCB_RLM_MK1 | REAL | 0.000000e+000 | 0.0
26.0 | Variables_SCB_RLM_MK1_TOLLERANCE | Variables_SCB_RLM_MK1_TOLLERANZA | REAL | 0.000000e+000 | 0.228
30.0 | Variables_SCB_RLM_MK2 | Variables_SCB_RLM_MK2 | REAL | 0.000000e+000 | 999.0
34.0 | Variables_SCB_RLM_MK2_TOLLERANCE | Variables_SCB_RLM_MK2_TOLLERANZA | REAL | 0.000000e+000 | 0.228
38.0 | Variables_SCB_RLM_MK3 | Variables_SCB_RLM_MK3 | REAL | 0.000000e+000 | 0.0
42.0 | Variables_SCB_RLM_MK3_TOLLERANCE | Variables_SCB_RLM_MK3_TOLLERANZA | REAL | 0.000000e+000 | 0.228
46.0 | Variables_SUM_Material_Use | Variables_SUMME_MATERIALEINSATZ_MM | REAL | 0.000000e+000 | 2512220.0
50.0 | Variables_SUM_Section_MM | Variables_SUMME_ANSCHNITT_MM | REAL | 0.000000e+000 | 14160.0
54.0 | Variables_SUM_Remaining_End | Variables_SUMME_RESTENDE_MM | REAL | 0.000000e+000 | 53693.82
58.0 | Variables_SUM_SPAENE_MM | Variables_SUMME_SPAENE_MM | REAL | 0.000000e+000 | 33149.5
62.0 | Variables_SUM_Workpieces_MM | Variables_SUMME_WERKSTUECKE_MM | REAL | 0.000000e+000 | 2411217.0

Here is my struct:

` public struct testStruct
    {
        public bool o1_Req_GUD;
        public bool o2_GUD_Error;
        public bool o3_GUD_NDR;

        public int o4_AbsoluteCounter;
        public int o5_TargetPieces;
        public int o6_Pieces;

        public byte o7_Length;
        public byte o8_Measured_Length;

        public byte o9_sensor1;
        public byte p10_sensor1_tolerance;
        public byte p11_sensor2;
        public byte p12_sensor2_tolerance;
        public byte p13_sensor3;
        public byte p14_sensor3_tolerance;   

        public byte p15_total_material_use;
        public byte p16_total_section;
        public byte p17_total_remaining_end;
        public byte p18_total_spaene_mm;
        public byte p19_total_work_pieces;
    }`
</body>

</html>

I think I might be using the wrong c# type for DINT.  Any ideas?  If someone is willing to work with me on this, I would gladly pay for the assistance.

bgreer5050, 2022-01-09:
Specifically o6 is producing a negative number.  It should never be negative which leads me to believe I need to use another data-type here.  If I use a byte[] I'm unsure how to convert it to something useful.

Jason-Jelks, 2022-01-10:
You need to consider the endianness of variable type in the Siemens PLC. Things like DINT, DWORD, LINT, REAL, LREAL... et al, are byte arrays, but they are Big Endian and need to be converted if you are just reading straight byte[]. The Class and Struct helper classes do this for you, but you will need to associate the correct size (bit count) 8-bit/16-bit/32-bit/64-bit.

Siemens             C#
BOOL                 bool
BYTE                   byte
WORD                short / ushort
DWORD              int / uint
INT                     short / ushort
DINT                   int / uint
REAL                   float
LREAL                 double

AtlanticDev, 2022-01-10:
I'm not sure I understand how to make sure I am associating the correct bit count
"_you will need to associate the correct size (bit count) 8-bit/16-bit/32-bit/64-bit._"

How would I go about just grabbing address 10 and 14 (separately) in my example above?  

   var variableInAddress10 = plc.ReadBytes(DataType.DataBlock, 1, 10, 4);
   var variableInAddress14 = plc.ReadBytes(DataType.DataBlock, 1, 14, 4);

Jason-Jelks, 2022-01-10:
In that case, you will be responsible for converting/concatenating the byte order since you are explicitly reading 4 consecutive bytes as a byte array. to properly convert the byte[] to an integer you would need to reverse the order of the bytes.

If you look at the Types (class) DInt.cs FromByteArray(byte[] bytes) method, you will see that the bytes have to be reversed (Big Endian) in order to match the value in the PLC.

scamille, 2022-01-10:
Oh and use ReadStruct and not ReadClass if you are definining your variables as member fields instead of properties.

Jason-Jelks, 2022-01-10:
I agree with @scamille . Use ReadClass instead of ReadBytes would be much easier

Jason-Jelks, 2022-01-10:
If you do use ReadClass, you will also need to make sure you have a Setter in your properties and a plain constructor.

AtlanticDev, 2022-01-12:
Is there a way that I can connect with TIA software and verify the datablock?
I am looking for someone that might be available to work with me on this issue.  If there are any contractors here, please reach 
out to me.  bgreer@sigmasouth.com

mycroes, 2022-01-12:
Dear Bill, I'd love to help you, but I'm a bit time-constrained at the moment. What's your deadline to get this working? I know the class / struct reading code is messy, but I'm sure we'll figure what's wrong.

Regards,
Michael