S7NetPlus/s7netplus #426: Browse tag list, determine type

vernturbine, 2022-01-18:
Hello,

Is there any way to browse the PLC variables and/or to determine the data type of a variable?

Thanks, Vern

mycroes, 2022-01-18:
AFAIK there isn't. I know you can list data blocks and function blocks and more, but I'm not aware of any way to read tag lists. Also, data is stored contigiously in data blocks, there's nothing there telling where a variable starts or what it's type is.

closed 2022-01-18 by mycroes

vernturbine, 2022-01-18:
Thanks for the quick response!

I will carry on with importing a tag list that the user would need to first export from TIA.

yanzixiang, 2022-01-20:
> Hello,
> 
> Is there any way to browse the PLC variables and/or to determine the data type of a variable?
> 
> Thanks, Vern

If you are using S7 1500,
There is some way.

Jason-Jelks, 2022-01-20:
There may be a way to do this using TIA Openness, but I have not tried it since I believe it is a licensed package from Siemens.