S7NetPlus/s7netplus #429: Connect PLC via a PN-PN-Coupler

nanotask, 2022-02-18:
Hey there, 

does anybody has experiences with a pn-pn-coupler and know how to handle it with s7netplus? 

In my case, I have a SINUMERIK  NCU 710 (integrated S7300) connected to a pn-pn-coupler (Siemens PN-PN-Koppler 6ES7 158-3ad10-0XA0 V4.2). If I connect my laptop directly to the PLC then I can open a connection. But unfortunately the network comprises a coupler.  How do I get the data blocks from the coupler?

Does anyone have a tip or suggested solution for this hardware setup?

btw I am a student and I have not much programming experience.

bolt-b, 2022-02-19:
From the top of my head...
**Physically**, and cheaply there's no way to do that, except for hanging on your side a configured plc. Then you physically get access to this physical plc. Then you will able to read coupler's memory areas.
**Virtually**, you can also hang on your side virtial plc(by means of plcsim), then get access to this virtual plc. Then you will able to read coupler's memory areas.

mycroes, 2022-02-24:
From what I'm reading PN-PN couplers don't offer datablock access at all. Also, they're using Profinet as communication protocol (hence the PN in the name), this library doesn't support Profinet in any way. I think you're out of luck unless you get direct access to the PLC.

closed 2022-02-24 by mycroes

yanzixiang, 2022-02-28:
[image]

yanzixiang, 2022-02-28:
virtial plc(by means of plcsim) does NOT support profinet.

> From the top of my head... **Physically** and cheaply there's no way to do that, except for hanging on your side a configured plc. Then you physically get access to this physical plc. Further you will able to read coupler's memory areas. **Virtually**, you can also hang on your side virtial plc(by means of plcsim), then get access to this virtual plc. Further you will able to read coupler's memory areas.