S7NetPlus/s7netplus #430: Read incorrect value when I put a float variable in a struct

funnyYang1994, 2022-03-22:
when I read a struct and write line a float type  variable, the value of that variable is "6.157642E-39" ,but the value in plc is "141.0".I  had searched document for long time ,but still have no idea. Could some one help me? 
Thank you very much!
[image]
[image]

mycroes, 2022-03-22:
Hi @funnyYang1994, Have you tried reading just a single value at the correct offset? My guess is the offset calculation in S7NetPlus is wrong, I would suggest trying to read the AxisConfig structs only starting at the correct offset. Please report back if that works.

The separate bits and bytes might be throwing off the calculation, I'm assuming that `testStruct` has 22 bytes for `codeHSG1`, 2 bytes for `spareBit1` to `spareBit3`, 2 bytes for `spareByte1` and `spareByte2` and so `oilAxisX` is starting at an offset of 28 within the struct. Can you confirm that as well?

scamille, 2022-03-22:
My assumption would be that oilAxisX starts at offset 26, but S7NetPlus might calculate it to offset 25, because it does not correctly pad the last byte.
0 + 22 (string) + 2 (3x bool, 1x byte), + 2 (1x byte, 1x byte padding) = 26

scamille, 2022-03-22:
Code example to test what @mycroes suggests:
`var oilAxisX = (AxisConfig)plc.ReadStruct(typeof(AxisConfig), 1750, 28);` or
`var oilAxisX = (AxisConfig)plc.ReadStruct(typeof(AxisConfig), 1750, 26);`

mycroes, 2022-03-22:
> My assumption would be that oilAxisX starts at offset 26, but S7NetPlus might calculate it to offset 25, because it does not correctly pad the last byte. 0 + 22 (string) + 2 (3x bool, 1x byte), + 2 (1x byte, 1x byte padding) = 26

I miscalculated (26 is what I was aiming for), I'm not sure if the byte is added after bits without padding though.

funnyYang1994, 2022-03-23:
> Hi @funnyYang1994, Have you tried reading just a single value at the correct offset? My guess is the offset calculation in S7NetPlus is wrong, I would suggest trying to read the AxisConfig structs only starting at the correct offset. Please report back if that works.
> 
> The separate bits and bytes might be throwing off the calculation, I'm assuming that `testStruct` has 22 bytes for `codeHSG1`, 2 bytes for `spareBit1` to `spareBit3`, 2 bytes for `spareByte1` and `spareByte2` and so `oilAxisX` is starting at an offset of 26 within the struct. Can you confirm that as well?

I check the size of the struct in plc , the wrong offset does cause this issue. The first image is the struct in plc ,I change the struct in C# program like the second picture , and now I can read value correctly. Thank you very much again!  
[image]
[image]

closed 2022-03-23 by funnyYang1994