S7NetPlus/s7netplus #432: S7.net PlcException

tjgk411857, 2022-05-20:
Hi guys
I use firstly S7netplus to get S7-1500 Data. I download the new version S7NetPlus. but I meet the following problem about this after PLC already connected. can you give me a advice.?
private void Btn_ReadBool_Click(object sender, EventArgs e)
        {
            // Get Siemens Bool Type
            bool Result1 = (bool)plc.Read("DB6.DBX32.0");
            Txb_BoolValue.Text = Result1.ToString();
        }

error message :S7.net PlcException:"Receive 12 bytes: "32-02-00-00-00-00-00-00-00-00-81-04", expected 19 bytes"

tjgk411857, 2022-05-20:
[image]
This is the PLC Struct .

easyboot, 2022-06-27:
我也出现了，这个问题，让PLC工程师看了一下，目前正常了

emoriver, 2022-06-27:
> 我也出现了，这个问题，让PLC工程师看了一下，目前正常了

What should look for PLC's engineer (which is me, actually)?

easyboot, 2022-07-15:
我自己安装了一个环境进行了测试
WIN10纯净版+博途V17+PLC1200

如果不勾选远程读写，则会报上面的错误，勾选之后就OK了

[image]