S7NetPlus/s7netplus #440: PLC: Address out of range.

Ahmed2017Saad, 2022-07-16:
Hello
I got this error each time I tried to write data to the last address of the Data block, the plc is s7-300

the address I tried to write to is DB63.DBW50, The data block length from the PLC is 52 bytes, and I can access this address from PLC successfully 

also, I tried to write to the last word of another data block I got the same error
 

S7.Net.PlcException
  HResult=0x80131500
  Message=Received error from PLC: Address out of range.
  Source=S7.Net
  StackTrace:
   at S7.Net.Plc.WriteBytesWithASingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] value, Int32 dataOffset, Int32 count)
   at S7.Net.Plc.WriteBytes(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] value)
   at S7.Net.Plc.Write(DataType dataType, Int32 db, Int32 startByteAdr, Object value, Int32 bitAdr)
   at S7.Net.Plc.Write(String variable, Object value)
   at WindowsFormsApp2.Form1.btnDelet_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at WindowsFormsApp2.Program.Main()

  This exception was originally thrown at this call stack:
    [External Code]

Inner Exception 1:
Exception: Received error from PLC: Address out of range.

mycroes, 2022-07-16:
Could it be you're writing an int while the plc type is word (short in C#)?

Ahmed2017Saad, 2022-07-16:
I  have tested it after converting the data to ushort, and it is working

but another issue noticed

i'm trying to write to DB63.DBB50 byt using
`plc.Write(DB63.DBB50, (ushort)10);`

when monitoring from PLC the byte is still zero as shown below
[image]

mycroes, 2022-07-16:
You should cast to byte instead of ushort when you want to write a byte.

Ahmed2017Saad, 2022-07-16:
i have updated the casting to byte, it wrote to the correct plc address but the data is wrong
i was trying to write 10
please check the Wireshark data and plc data   please check the address **DB63.DBB0** it is 2 instead of 10

[image]

[image]

Ahmed2017Saad, 2022-07-17:
sorry for my mistake, the plc was writing at the same time to the same address, that is what made the data changes

closed 2022-07-17 by Ahmed2017Saad