S7NetPlus/s7netplus #441: S7-1200 Reading Datablocks - WrongVariableFormat

mustafaxfe, 2022-07-28:
I have been following this instruction ( https://www.youtube.com/watch?v=tYTjNG8YL-c ) to read data blocks.
It works normally using PLCSIM (NetPLCSim). 

But It is not working with my PLC hardware.

It returns "WrongVarFormat" exception for every Read call. I can read the value of DB2.DBX0.0 addressed with PLCSIM but it is not working with PLC(S7 1200) hardware.

I made these configurations but it didn't solve my issue.

- Optimized block access disabled
- Protection - Full Access Protection
- Permit Access with PUT/GET from remote partner.

bolt-b, 2022-08-08:
hi,
could you show me a piece of the code (C#), also it would be nice to see your DB in TIA Portal that you are trying to read (make a picture of that).

mustafaxfe, 2022-08-22:
I made a factory reset and this solved my issue. Everything works as intended. Thank you.

closed 2022-08-22 by mustafaxfe