S7NetPlus/s7netplus #445: Using s7netplus in ASP.NET Core3.1 Blazor Server Side App.

mut9bu, 2022-09-18:
Hello 
I wanted to user you dll in my Blazor Server Side App. (ASP.NET Core3.1).
1) I have downloaded and installed the nugget 0.14.0. (No Error during installation)
2) I have downloaded and copied the complete "s7netplus-develop" directory in my project directory where my *.sln is
3) Then I wanted to add reference to the dll file. (I just found one dll in "S7.Net.UnitTest\runtimes\win-x64\native\Snap7.dll" directory.
    ==> If I tried to add reference I ger ERROR "The reference is invalid or unsupported"
Did I something wrong or is my ASP.NET Core version not fitting?

mut9bu, 2022-09-18:
Have done it. The S7Net.dll was already installed with ne nugget. I just gave reference Using S7.Net

closed 2022-09-18 by mut9bu

peterpanBest, 2022-11-28:
> Have done it. The S7Net.dll was already installed with ne nugget. I just gave reference Using S7.Net
Hello, I try to use s7netplus in .net core3.1 but  it did't work well. when you deploy project to IIS, it did not work. you should use [Sally7](https://github.com/mycroes/Sally7) if you use S7 by .net core3.1. i hope this advise could help .