S7NetPlus/s7netplus #446: Is it possible to read all the values from a DB from aS7-300 PLC without knowing the structure?

mut9bu, 2022-09-18:
Hello
With S7.Net I want to monitor data especially from DB's of S7-300 Controllers.
But the problem is that I don't know the structure for each DB of each machine.
I want to write a code that reads first the structure of the DB and accordingly all the data in the structure.

With other words: How can I see all the data in a Data Block with the correct structure,  without defining the DB structure in my code?
Is that possible?

yanzixiang, 2022-09-21:
no
it is not stored on the plc,
it is stored in the project on the pc.

mut9bu, 2022-09-21:
Thanks for the answer, I know that the neccesary information for the DB structures and symbols are in a SUBBLK.DBT fiel in the S7-Project. I wanted to read them out from this file, but some part of the file seems to be binary. Especialy the part between the structure information. Do you know how to read out the DB structures correctly from this file?