S7NetPlus/s7netplus #450: PLC POLLING STOP TO UPDATE MY CLASS PROPERTIES.. WHY???

LucaPannozzo, 2022-10-30:
Hello everyone,

I developed a Consolle App in c#. The app polling 7 siemens s7 1200. during the polling i update my classes properties and waiting some bits switch true for execute my job and the write a feedback bit to the plcs. I build a polling class like these:

`

    public abstract class ServizioPlc
    {

        public EventHandler OnValuesRefreshed;

        public EventHandler ReadValues;

        public EventHandler WriteValues;

        public string Port { get; private set; }
        public string IpAddress { get; private set; }
        
        internal System.Timers.Timer PollingTimer { get; set; }

        internal DateTime LastScan { get; set; }

        public TimeSpan ScanTime { get; set; }

        public bool ReadError { get; private set; }
        public bool WriteError { get; set; }
        public string LastError { get; internal set; }
        public TimeSpan WriteTime { get; private set; }
        public TimeSpan ReadTime { get; private set; }
        public TimeSpan UpdateTime { get; private set; }
        public EventHandler OnWriteError { get; set; }
        public EventHandler OnReadError { get; set; }

        public ServizioPlc(string ipaddress, string port)
        {
            IpAddress = ipaddress;
            Port = port;

            PollingTimer = new System.Timers.Timer();
            PollingTimer.Interval = 1000;
            PollingTimer.AutoReset = true;
            PollingTimer.Elapsed += PollingTimer_Elapsed;
        }

        public void EseguiCicloPolling()
        {
            PollingTimer_Elapsed(new object(), null);
        }

        internal void PollingTimer_Elapsed(object sender, ElapsedEventArgs e)
        {
            //PollingTimer.Stop();
            
            try
            {
                Execute();
            }
            catch(Exception ex)
            {
                Console.WriteLine("PLC SERVICE "+IpAddress+"- Eccezione non gestita durante esecuzione Pollig. " +ex.Message + ex.InnerException?.Message + ex.StackTrace + ex.InnerException?.StackTrace);
                
                if (OnReadError != null)
                    OnReadError.Invoke(ex, new EventArgs());
            }
            
            ScanTime = DateTime.Now - LastScan;
            LastScan = DateTime.Now;

            //PollingTimer.Start();
        }
[... 226 lines cut ...]
        internal void FeedbackPalletTerminato(object sender, EventArgs e)
        {
            if (!Client.Plc.IsConnected)
                Client.Plc.Open();

            Client.Plc.WriteBit(S7.Net.DataType.DataBlock, 5010, 1, 0, true);

        }

        internal void ResettaFeedBackPalletTerminato(object sender, EventArgs e)
        {
            if (!Client.Plc.IsConnected)
                Client.Plc.Open();

            Client.Plc.WriteBit(S7.Net.DataType.DataBlock, 5010, 1, 0, false);
        }
    }
}
`

NOW, the main problems is, after a long periodo maybe one hour, my classes no update the properties with plc values and i need to restart the application. After do that, system start to works properly. i suppose the problem is with connection /socket the application/GC dispose something or i don't know. I'm not a .NET expert so probably I'm not write correctly the code but with Omron Drivers works well, only with s7.net I have these issues! 
Please, someone understand what is happening?

Best regards everyone 
Thank you

Jason-Jelks, 2022-10-31:
There are some factors that could have an effect on this. Depending on the interval/how often you are opening and closing the connection to the PLC, you could be consuming multiple connections internally into the PLC. From your code, it is possible you are opening "new" instances, but have not necessarily closed and disposed of previous ones.

The Siemens S7-1200 has a maximum number of external connections available, and once those are reached, it cannot accept any more. When you terminate your program, the TCP Sockets itself will terminate and release in the Siemens. Again, this is merely a possibility. If you have access to the Siemens PLC via TIA Portal, you can monitor the number of connections taken and available to see if this is actually occurring. 

As an aside, you could make your PLC object a factory and hold the instance connection open and then close it only when you dispose of the factory instance. This would be a cleaner approach.

mycroes, 2022-10-31:
I used to believe you can actually only have a single connection open for the specified local and remote TSAP, but I recently had a situation where multiple connections using the default TSAP actually seemed to work. Nevertheless @Jason-Jelks is absolutely right, you should manage your connection in whatever way to ensure they're not dangling.