S7NetPlus/s7netplus #451: Select Hardware

OscarOrefice, 2022-11-01:
https://github.com/S7NetPlus/s7netplus/blob/7f76d4fc5a840a52021f766d749f4cef39e1d675/S7.Net/PlcAsynchronous.cs#L49

I'm not a contributor but I propose to add the endpoint direct on the ethernet board, I use one panel PC with 3 ethernet and some times  the local area address of customer it's in contrast with my machine area, if is possible to modify upper row with this suggest

TcpClient c=new TcpClient(new IPEndPoint(IPAddress.Parse("192.168.10.250"), 0);

scamille, 2022-11-02:
It might be best to allow injecting a TcpClient factory. That way the user of the Plc class can decide how the TcpClient gets constructed if he needs to - like in your case.

OscarOrefice, 2022-11-02:
maybe passing only the endpoint as a parameter, a small change but that adds a mode to the method