S7NetPlus/s7netplus #452: from souce code,its should be get exception,but actually is successful

maxwelltsai1536, 2022-11-11:
the PlcAsynchronous.cs line 458 method ValidateResponseCode , i saw the s7data[14] is 0x00,here should be throw exception ,but acctually not,this statusCode is Reserved
[image]

mycroes, 2022-11-11:
Do you experience an actual issue, or are you guessing something must be wrong? People are using this library, I would assume someone would've noticed if this was broken.

maxwelltsai1536, 2022-11-11:
> Do you experience an actual issue, or are you guessing something must be wrong? People are using this library, I would assume someone would've noticed if this was broken.

Do you experience an actual issue -> No
are you guessing something must be wrong?  -> Yes （from souce code logic,i guess it should be wrong,but actually is successful,this is where i get confused）

mycroes, 2022-11-11:
My guess is the indexes don't align because S7NetPlus is performing this check on a subset of the returned data. You can probably easily detect that by looking at all of the data. Also, in [Sally7](/mycroes/sally7) most of the messaging structure is available as structs, that might provide an easier look into the protocol data from C# perspective.

maxwelltsai1536, 2022-11-11:
> My guess is the indexes don't align because S7NetPlus is performing this check on a subset of the returned data. You can probably easily detect that by looking at all of the data. Also, in [...]

Got it.i will check agin .thank you for reply.

closed 2022-11-14 by maxwelltsai1536