S7NetPlus/s7netplus #453: Can't read a single Real

Wobeam, 2022-11-11:
Hello,
`
float plc_data;

plc_data = (float)plc.Read("DB1300.DBD0.0");
`
In PLC DB shown absolute address is Real type.

SmackyPappelroy, 2022-11-11:
You write 
plc_data = (float)plc.Read("DB1300.DBD0");

Wobeam, 2022-11-12:
> You write plc_data = (float)plc.Read("DB1300.DBD0");

Thanks of cource but it doesn't matter.

rubend30, 2022-12-16:
Use the `Read(DataType dataType, int db, int startByteAdr, VarType varType, int varCount, byte bitAdr = 0)` method instead if you want the correct type in one call.

As seen in the internal `PLCAddress `class a `DBD `address is always converted to a `DWord` and that translates to a C# ` uint`:

`case "DBD":
 varType = VarType.DWord;
 return;`

blackhole997, 2022-12-29:
Just do this:
`var floatNumber = BitConverter.ToSingle(BitConverter.GetBytes((uint)this.LibPlc.Read("DB200.DBD34")), 0)`

FalcoGoodbody, 2023-02-20:
You can also use the method which is described in the wiki at "value conversion":

```
float result = ((uint)plc.Read("DB1.DBD40")).ConvertToFloat();
```

Wobeam, 2023-02-21:
> You can also use the method which is described in the wiki at "value conversion":
> 
> ```
> float result = ((uint)plc.Read("DB1.DBD40")).ConvertToFloat();
> ```
> 
> OR you use the method to read multiple vars from a list. The list can also contain only one item. When declaring the items you can define the datatype.

Yeah, found conversion in docs last year. Thanks.