S7NetPlus/s7netplus #454: Reading named pipe on Linux

hrvojej, 2022-11-23:
Hi,
PLC newbie here, bear with me please. 
I've build PLC server and client for testing on Linux/Ubunut 22.04 and all works fine on Ubuntu.
I'm able to run server and write to named pipes with your test client ok. 
Server displays that it's correctly changing status of named pipe. 

Now I want to do replace your test client with basic Java app that needs to write to pipe with something like :

```
RandomAccessFile pipe = new RandomAccessFile("\\\\.\\pipe\\pipename ", "rw");
                short sVal = 1;
                pipe.write((byte) sVal);
```
I googled all over but for starters I'm unable to locate where this named pipe lives in my Ubuntu?    
So this part is troubling me `"\\\\.\\pipe\\pipename "  `     
This is probably incorrect but I don't understand how to find and referece named pipe in Java.

mycroes, 2022-11-23:
HI @hrvojej,

I enjoyed reading your question 😄 I'm not a PLC expert either, but I'm also not a Java expert. Actually, I don't consider myself a Linux expert either, and S7NetPlus doesn't actually focus on either of the 3 technologies at all. However, I guess I still have the answer to your question.

I think named pipes are a Windows only feature. I'm not sure about their exact implementation, but I believe they're actually quit similar to socket files in Linux. I'm not sure if `RandomAccessFile` actually requires an existing socket file, but either way it needs to point to a file system path. Possibly `\\\\.\\pipe\\pipename ` actually works (note that you also have a space at the end of the name here, that can lead to annoying mistakes), but it's not really convenient. If it does work there'll be a file in the current working directory with that name. You probably want to use either a file in `/tmp` or `/run` if you want to adhere to common Linux behavior.

Anyway, I hope this gets you on your way. If it doesn't, I'd suggest searching StackOverflow for a more specific issue of what you're running into, because this currently isn't related to S7NetPlus in any way. As such I'm also closing this, but if you run into any issues that relate to S7NetPlus feel free to open a new one.

Regards,
Michael

closed 2022-11-23 by mycroes

hrvojej, 2022-11-23:
Ok, thanks!