S7NetPlus/s7netplus #456: Is It possible to listen from PC data from PLC and then do something?

ComptonAlvaro, 2022-11-27:
Hello:

I would like to know if it is possible that the PLC send data to a PC that is listening. In case the the PC receive the data, it can anylize and act in consecuences. For example, if the PLC send an alarm, the PC could send an email or save information in a database.

If I am not wrong, all the examples in the documentation is the case in which the PC read the data from the PLC, so the PC has to request the data to the PLC. But in my case I would like that would be the PLC who send the data and the PC just to wait.

If not, the alternative would be  that the PC had to read periodically from the PLC, and I guess it is a little worse solution.

Thanks.

yanzixiang, 2022-11-28:
you can use ladder program or other to write some program sending TCP OR UDP packet out,
and write some program on the PC waitting for that packet,
this is called both side programe.

PC reads periodically from the PLC,
PLC just defines the db,
this is called single side programe,

I don't think all programers here know how to write PLC programs,
so single side programe is better.

scamille, 2022-11-28:
If you go the route of periodically polling the PLC from the PC side, you can also define some custom bytes indicating eg. "new data available" and even "successfully read new data".

You can imagine the PC side polling just the "new data available" byte. If 0, it just continues. If 1, it then reads the full DB.
Once the PC has successfully read & saved the full DB (to a file, database, whatever), you can then set the "successfully read new data" byte, letting the PLC know that you have read the DB.

The PLC can then also implement logic based on these 2 bytes. You can for example think about what should happen if the PC did not yet read the previous data, but the PLC already has new values available. You could just overwrite the previous data, or have a ring-buffer allowing for some outage of the polling PC. (Depending on your need, and the amount of memory you are willing to sacrifice for this).

mycroes, 2022-11-28:
> @scamille using a byte for that it could be a possible, but from my point of view it has the same base disadvantage, the PC has to polling the PLC periodically to know if there are new events or not.

It also has an advantage. If the update is one-time only and it gets lost in whatever way it's lost forever. If PC polls the signalling byte and it misses a read or increment it'll catch the next one.

> Also I have to control the concurrency, what happen if there are new events between the PC get the new data and just bejore set to 0 the byte, the PLC add new events?

Exactly, *you* have to *control* the concurrency. There's many books on concurrent systems, you can probably consult any one of them for multiple solutions to this problem.

> It is why I prefer the PLC send the data to the PC, which is waiting for events.

That's fine, but it's out of scope for S7NetPlus.

> @yanzixiang Although here perhaps most of the people don't know how to implement both side programe, perhaps the first question would be, is it possible the PLC notify the event to the PC or it is not possible and the PC has to read periodically the avlues of the PLC?

I think this has been answered, programming a TCP or UDP client on the PLC and the counterpart on PC can achieve this, it's not possible in S7 scheme of things.

All in all I recommend polling, but of course you could be dealing with way more data than I am. I'm closing this issue because it doesn't apply to S7NetPlus itself, but feel free to continue the discussion.

closed 2022-11-28 by mycroes

ComptonAlvaro, 2022-11-28:
@mycroes Thanks for your comments.

I have a doubt, when you tell there are many books that expose solutions for the concurrency control, do you mean in a .NET environment or between PLC and PC? I know how to control concurrency in a C# application, but I am not sure the best way to control concurrency between PLC and a PC application. Could you recommend some book for that case?

Thanks.

mycroes, 2022-11-28:
I meant about the principles of concurrency, because the principle apply to any device or language. I personally have this book: [Distributed Systems](https://www.distributed-systems.net/index.php/books/ds3/) (older edition though) and I found it a very good book on explaining distributed systems and concurrency. I had that book for my Computer Sciences study, I definitely recommend it.

Also, what we're basically using at my work is the following:

- PLC Datablock with status and additional data fields
- PC Datablock with command and additional data fields

In case of PLC initiated protocol:
- PLC sets data fields and status to 1
- PC sets response data and sets command to 1
- PLC starts processing and sets status to 2
- PLC finishes processing and sets status to 3
- PC confirms finish with command 3
- PLC returns to status 0
- PC returns to status 0

Keep in mind that:
- PLC will need to write data first, then status
- PC will need to read status first, then data (can be in a single call, but must be in this order to avoid reading old data and new status)
- PC will need to write data first, command last (again, can be a single call, but must be in this order)
- PLC reads command first, then data

You're not limited to the values I used above, but if the PC command will only toggle between 0 and 1 you can lose synchronization. Also keep in mind that PLC should wait for the corresponding PC command before it continues processing.

I'm not saying this is the way to go, it's just one possible solution. It's also something that both developing parties (PLC and PC) need to agree on, but we're quite happy with this approach. Even then you can choose whether to use this as a signaling mechanism only or whether PLC will stay in status 2 for the duration of the process, but that's up to you.

I hope you can find use for any of the pieces of information here.

ComptonAlvaro, 2022-11-28:
Thanks so much.

scamille, 2022-11-28:
> Also I have to control the concurrency, what happen if there are new events between the PC get the new data and just bejore set to 0 the byte, the PLC add new events?
> 
> It is why I prefer the PLC send the data to the PC, which is waiting for events.
> 

You still have exactly the same problem the other way around.

Now the PLC has to make sure that the data is sent to the PC. What if the PC isn't listening to TCP connection requests? What if it is too slow or has some other issue?

You have to decide what to do what happens if you get new data before he previous one is successfully sent to the PC in either case :-)

ComptonAlvaro, 2022-11-28:
That's true that I should to control from the PLC that the data was saved in the server, but this is not a concurrency problem, because the only who will save (and handle the data) is the PLC, There is no another process that will try to manipulate the same data.

It is true that it is possible that if the PLC tries to saved the data in the server, the network can down between when the server get the command to save and when the server will notify the correct operation. But in this case, I still think it is more easy to handle. If the server doesn't receive the notification from the server, it will try in the future. If the data was saved it will receive an error of unique and then only it is needed to delete the data.

If is the PC who polls the PLC, then there are two processes that will modify or handle the same data. In this case, it is needed to handle concurrency.

Of course, the PC has to control what happen if the network or PLC is not available since it get the data and tries to update the data in the PLC deleting the saved alarts? It is a common problem.