S7NetPlus/s7netplus #461: PLCSIM Advanced - Simulator Transfer error

paviano, 2023-01-13:
Hi,
I want to report that apperently Read/Write doesn't work while working on PLC SIM Advanced V4
The PLC in use is S1500, correcly run on physical one.  Down below a screenshot of exception:

[image]

Any suggestion?

jiriurban21, 2023-03-20:
Check, if you have this option checked in PLC hardware configuration. 

[image]

paviano, 2023-03-20:
Yes, it is

closed 2023-08-09 by paviano

mycroes, 2023-08-10:
@paviano could you add the solution, so if someone else runs into the problem they know how to fix it?

@jiriurban21 thanks for chiming in!