S7NetPlus/s7netplus #463: Connection between plc and unity

TebyanAli36, 2023-02-04:
i do plc connection from the docummentation it work but then suddenly stpod messeging the error :
plcExecption: plc is not connected.

scamille, 2023-02-05:
You should check if the plc object is still connected before performing some operations (read/write).

The plc object is automatically disconnected if there previously was a non-recoverable exception during communication.

TebyanAli36, 2023-02-07:
then how can i check this??

scamille, 2023-02-07:
https://github.com/S7NetPlus/s7netplus/blob/6aa0133081048131aa269d0d5b780ec7fe01028f/S7.Net/PLC.cs#L114

TebyanAli36, 2023-02-09:
i have s7.net.dll library and just import it in unity firstly it work then start to give error (plc is  not connected) , and when i choose the ip address of the network in nettoplcsim appear two ethernet address ?