S7NetPlus/s7netplus #470: S7.Net.PlcException : Index was outside the bounds of the array.

OlivierMiracle, 2023-03-07:
Hi!
I am trying to get this to work, but I am getting the same exception. There are the steps i took:

- Create a global DB [DB1]
- Unchecked "Optimized block access" in its properties
- In properties of PLC (Simatic S7-1200) I have set full access level
- I have checked PUT-GET communication

Next, I have created a few unit test methods:

For checking avaibility:
```
[Fact]
public void PlcRead_IsAvailable()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);

    //Act
    bool result = plc.IsAvailable;

    //Assert
    Assert.True(result);
}
```

For checking connection:
```
[Fact]
public void PlcRead_IsConnected()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);

    //Act
    plc.Open();
    bool result = plc.IsConnected;
    plc.Close();

    //Assert
    Assert.True(result);
}
```

And, finally, for write and read instructions.
```
[Fact]
public void PlcRead_WriteAndRead()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);
    byte expected = 10;
    byte result;

    //Act
    plc.Open();
    plc.Write("DB1.DBB0", expected);

    result = (byte)plc.Read("DB1.DBB0");

    //Assert
    Assert.Equal(expected, result);
}
```

Only the last one is failing with exception:
```
S7.Net.PlcException : Index was outside the bounds of the array.
---- System.IndexOutOfRangeException : Index was outside the bounds of the array.
```

The problem doesn't not depend on type of variables, their number, casting etc.

kristjanp28, 2023-03-08:
Hello,
I have 1500 PLC but it should work the same. I will also try on 1200.
Was having same problems and I checked PUT/GET communication in "Protection & Security" settings. 

Don't forget to go offline and download hardware changes! Can be a small thing but easy to forget.

OlivierMiracle, 2023-03-08:
Ok, I needed to do what you said. Making changes in hardware settings and then clicking "download to device" seems to not change parameters in PLC itself.

To do that properly I had to choose PLC_1(my plc) > Download to device > Hardware configuration

I am overly descriptive for other lost souls like me.

closed 2023-03-08 by OlivierMiracle