S7NetPlus/s7netplus #471: Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host.

zoaaz21, 2023-03-18:
Hi, I'm trying to connect to a S7-300 and I'm receiving this error message.

This is my code, i try to replace IsConnected with IsAvailable but i receive the same problem

[image]

mycroes, 2023-03-18:
I know I'd get this error with S7 1200 / 1500 when the TSAP's are invalid, they are calculated using rack and slot number I think, so perhaps your error is there?

SmackyPappelroy, 2023-03-21:
Some CP-modules have a limit of 8 simultaneous connections. You can check in the diagnostics how many active connections the plc has

zoaaz21, 2023-03-21:
[image]

[image]
I have this configuration in the cpus that I can connect and i try to replicate this config but i have the same error

SmackyPappelroy, 2023-03-21:
Try disabling the firewall/antivirus on you computer. Port 102 must be open! Can you ping the CP-module?

closed 2025-01-06 by zoaaz21