S7NetPlus/s7netplus #475: Subscription to the tags, and firing event for it

DonatelloLTU, 2023-04-17:
I have been working with different types of PLC and C# code for a while now. Found many different great libraries to achieve direct communication to PLC (including this library) rather than OPC server or so. I see one issue with this library compared to the others, is that this library does not have subscription to tag functionality. Basically where if value of the tag changes, it fires event, where then you map it to your own methods.
Did anybody ever tried to implement something like that using this library? 
When asked to chatGPT, they give me example of subscribing to the tag for this library, which I believe was previous owner version, but I might be wrong.
Thanks!

mycroes, 2023-04-17:
Hi,

Asking ChatGPT these things is not the best idea. Apparently ChatGPT thinks everything S7 related is the same thing, so it might as well have been based on a JavaScript library (not kidding). Try asking ChatGPT who the author of Sally7 is, last time I tried it came up with a name I don't know and when asking ChatGPT if he's sure it's not Michael Croes (me, the right answer) it'll keep coming up with other names.

Anyway, to answer your question, there are a few issues:
- We can't address variables by name
- We can't subscribe to changes

I believe it's possible to address variables by name, but not with the protocol version S7NetPlus is using. I'm also not aware of any library that can do addressing by name, if you can point me to one I'm willing to investigate the possibilities.

As for the second point, I believe there has been support for cyclic reads in older S7 PLC's (the 300 for instance), but it's not available on newer PLC's, AFAIK. It's not hard to poll a variable for changes though. I don't necessarily think that should be part of S7NetPlus though. Professionally I'm sharing my polling behavior over multiple PLC connection libraries, I'd prefer that over a polling implementation per PLC...

DonatelloLTU, 2023-04-18:
Michael,

I understand that asking chatgpt such things is not the best idea, especially with limited time frame knowledge, though whenever I did, it showed all the methods as currently within the library (new Plc; .Open(....); .Read(....) and so on) besides the .Subscribe() one. Could have been that took some other library and combined with s7net. That is the reason it took me on this path to check with you.

Sorry for misunderstanding about the tag names. What I meant was address of the tag rather than the name. Been working with too many different PLCs at the moment.

I was thinking to implement polling, but wanted to check with you guys first is there any built in function within the library. 

Appreciate your answers!

closed 2023-04-18 by DonatelloLTU