S7NetPlus/s7netplus #476: Missing Types at Serialization

TobyStoe, 2023-04-26:
I tried to use the .Write Method with different Datatypes. It seems like there are missing some Datatypes for the Serialization.

S7.Net.Protocol.Serialization.SerializeValue

Missing Types are:
 [S7]/[C#]
-sInt/sByte
-LInt/Int64
-ULInt/UInt64
+ the Array-Version of these 3

mycroes, 2023-04-28:
Yes, these are indeed missing. I'm quite time constrained so I won't make any promises about adding them, shouldn't be too hard to add them similar to the current implementations though.

paviano, 2023-05-19:
Either a System.Boolean[]/BitArray is not supported yet. Especially to write an entire array it could be very useful. 
[From Stackoverflow]

```
internal static byte[] BoolsToByteArray(bool[] bools)
    {
        int len = bools.Length;
		int bytes = len >> 3;

		if ((len & 0x07) != 0) 
			++bytes;

        byte[] arr2 = new byte[bytes];

        for (int i = 0; i < bools.Length; i++)
        {
            if (bools[i])
                arr2[i >> 3] |= (byte)(1 << (i & 0x07));
        }

		return arr2;
    }
```