S7NetPlus/s7netplus #484: Cannot catch any InvalidAddressException from PLCExceptions.cs

dylandrush, 2023-06-16:
When using `DataItem.FromAddress(addr)`, it calls `PLCAddress.Parse` which can throw the exception `InvalidAddressException`.  Either the exceptions from PLCExcpetions.cs should be public, or in this case,  `DataItem.FromAddress` could wrap PLCAddress.Parse with a try/catch:

```
public static DataItem FromAddress(string address)
{
    try
    {
        PLCAddress.Parse(address, out var dataType, out var dbNumber, out var varType, out var startByte,
            out var bitNumber);
    }
    catch (InvalidAddressException ex) {
        throw new PlcException(ErrorCode.WrongVarFormat, ex);
    }

    return new DataItem
    {
        DataType = dataType,
        DB = dbNumber,
        VarType = varType,
        StartByteAdr = startByte,
        BitAdr = (byte) (bitNumber == -1 ? 0 : bitNumber)
    };
}
```

mycroes, 2023-06-17:
I guess they should all be public, I'm not even sure why they aren't in the first place... Feel free to submit a PR, otherwise I might be taking it up soon.

closed 2023-06-19 by dylandrush

mycroes, 2023-06-26:
Sorry for the delay, just released v0.19.0 with this change to NuGet.