S7NetPlus/s7netplus #486: s71500 write bit M9.2 How to read and write?

anan1213095357, 2023-06-26:
s71500 write bit M9.2 How to read and write?

mycroes, 2023-06-26:
What about `plc.Read("M9.2")`? I have my doubts looking at the code for converting the resulting data to `bool`, but this should be the way to read. As for writing, `plc.Write("M9.2", true)` should be enough, however I won't make any promises that it actually works.

I don't have a PLC to test this against and I have never used memory bits, but the above should get you going.

anan1213095357, 2023-07-30:
Yes, after I write this way, it's invalid

anan1213095357, 2024-03-31:
> Please provide some more detail:
> 
> * Provide code that reproduces this issue
> * Explain what happens
> * Explain what would've been the expected behavior
> 
> I'm assuming you're using something along the lines of what I provided above, but please provide an exact sample that causes your issue.

_plc.WriteBit(S7.Net.DataType.Memory, 9, 0, 1, true);    I read normally and write abnormally

HY-Liang-95, 2025-03-21:
_plc.WriteBit(S7.Net.DataType.Memory,  0, 9,  0, true); 试试这个