S7NetPlus/s7netplus #509: 在写入值时出现写入对象偏移,原本应该将结果写入第一个地址的自动顺延到了第二个地址里

2287728768, 2023-09-12:


mycroes, 2023-09-12:
Please provide an English description of your problem and add a minimal reproduction that shows when the issue happens.

2287728768, 2023-09-12:
When writing a value, there was a write object offset. The result that was supposed to be written to the first address was automatically postponed to the second address. How to solve this problem.

2287728768, 2023-09-12:
The DB address you want to write is DB1. DBW0.0, and the result is written to DB1. DBW2.0

mycroes, 2023-09-12:
Please add a minimal reproduction. The behavior you're experiencing doesn't seem to be a known issue.

2287728768, 2023-09-12:
yes,Write value offset by 2 bytes

mycroes, 2023-09-12:
Please add a minimal reproduction (code sample) that causes this problem.

2287728768, 2023-09-12:
Plc PLC20 = new Plc(CpuType.S71500, "192.168.12.254", 0, 1);
 PLC20.Open();
PLC20.Write(textBox2.Text, int.Parse(textBox1.Text));            //textBox2 To obtain the DB block address

2287728768, 2023-09-12:
The operating environment is Visual Studio 2017. 
PLC operates in a simulation environment
Thank you

2287728768, 2023-09-12:
such is the case. Thank you for explaining

closed 2023-09-12 by mycroes