S7NetPlus/s7netplus #515: Could be possible to read PLC's Metadata?

mendozagit, 2023-11-02:
By metadata, I mean.

- Variable names,
- Variable addresses, 
- Variable data types
- Others

gfoidl, 2023-11-03:
In general that's not possible, as in the PLC it's just a big block of bytes, and how they're addressed / interpreted is up to the user (or program). Similar to RAM on CPUs, where the metadata isn't stored along the data. It's up to the program on how to interpret that values.

Something like this would be possible with [optimized blocks](https://support.industry.siemens.com/forum/at/en/posts/optimized-and-non-optimized-data-blocks-and-cpu-cycle-time/155671) (for S7-1200 onwards), but S7NetPlus can't handle these operations, as Siemens has proprietary disclosure of the schema that is applied in that case.

See also https://github.com/S7NetPlus/s7netplus/issues/242

mendozagit, 2023-11-03:
@gfoidl, Thanks for your guidance! Good to know.

closed 2023-11-06 by gfoidl