S7NetPlus/s7netplus #519: Reading String from Siemens Plc and dispplay that in unity object text

JadoonJee, 2024-01-09:
Hi Everyone
I am having a problem with reading string from Siemens S71500 PLC, I am actually not sure how to read that in unity as a  string.
What I want to do is read a string name and then I want to display that in unity TextMashPro.
I have written something like this but no luck, I am getting a strange message like this **System.Byte[]**
 My Db number is 647 and  starting address is 31090 which is define as string and reading first 40 bytes, and using S7.net library
In a debug result of unity i get **System.Byte[]**
Can someone please help me out.

                    var bytes = plc.ReadBytes(DataType.DataBlock, 647, 31090, 40);
                    string numb1 = Convert.ToString(bytes);
                    Debug.Log(numb1);

Kind Regards

[image]
[image]
[image]

LBCQ123, 2024-01-10:
//Try to use this
bytes = plc.SieMens.ReadBytes(mType, (int)mAddr.mDBNumber, (int)mAddr.mDBOffset, MaxLenth + 2);
var gbk = Encoding.GetEncoding("GBK");
ret = gbk.GetString(bytes, 2, bytes[1]);

JadoonJee, 2024-01-10:
> //Try to use this bytes = plc.SieMens.ReadBytes(mType, (int)mAddr.mDBNumber, (int)mAddr.mDBOffset, MaxLenth + 2); var gbk = Encoding.GetEncoding("GBK"); ret = gbk.GetString(bytes, 2, bytes[1]);

Will try that one thank you very much