S7NetPlus/s7netplus #522: Will there be new nuget package?

jk987, 2024-01-31:
Hello,
last nuget package here https://www.nuget.org/packages/S7netplus is from August 2023.
Could you please release new package version?
I'm looking forward to ReadClock/WriteClock.
Can I help you with this somehow?

jk987, 2024-01-31:
Well, I didn't want to wait for nuget so I downloaded source codes and included them into my solution.
And ReadClock() throws exception "Response from PLC indicates error 0x8104."
My PLC is S7-1200.
I'll investigate further.

jk987, 2024-01-31:
In Wireshark it looks like this.

Request:
[image]

Response:
[image]

I would like to catch how TIA portal is doing it.

SmackyPappelroy, 2024-01-31:
We never got it to work on a S7-1200/S7-1500 only on S7-300 CPU, so that's probably why

andmattia, 2024-02-03:
I usually work with S7 1200/1500 and Virtual. Are you sure you use the correct slot/rack anc plc type in connection process?

jk987, 2024-02-03:
@andmattia 
>  Are you sure you use the correct slot/rack anc plc type in connection process?

Actually I wasn't interested in rack/slot because standard read and write operations of DBs worked pretty well. (?)
Please, do you really have ReadClock()/WriteClock() working on S7-1200/1500?

(Meanwhile PLC programmer prepared set of registers for this clock setting/reading so currently it isn't issue for me. But why don't make it better way.)

SmackyPappelroy, 2024-02-03:
Ot doesnt work on S71200/S71500, read my last reply. What dont you understand?