S7NetPlus/s7netplus #523: Isconnected issue

albasha007, 2024-02-02:
When I run
 myPlc1500.Open();
myPlc1500.IsConnected();

then i run
myPlc1500.IsConnected();

it  always reports True even though the PLC is disconnected.

i want a way to check the plc if still available or not 
any one can help me please

xxxxiaohuiiii, 2024-04-28:
I'd like to know what to do too. I have the same problem.

mycroes, 2024-04-28:
I think this has been pointed out before, but I'll repeat it here.

You can't rely on connection state. The `Plc.IsConnected` property directly returns the `TcpClient.IsConnected` property. This property only reflects if the last TCP message has been delivered successfully. The shortcoming to this approach is that the TCP connection might still be active while the PLC won't actually accept any messages, but in general I think the PLC closes the connection when it reaches the point it won't perform any further communication (for instance when TSAP identifiers don't match).

If you want to know if the PLC responds, just perform the action you wanted to do (i.e., perform the read you intended to do). That of course might result in exceptions as well (i.e. reading an invalid address), but checking the connection state first and then invoking whatever method you wanted to invoke will yield the same result.

If the above isn't suitable (i.e., your application relies on user actions and you want to check if the connection is OK on an interval) you can call `Plc.ReadStatusAsync()` or `Plc.ReadStatus()`, I guess these should always work without exceptions as long as the connection is OK. On the other hand, if they don't work, I can't imagine that reading or writing data is a possibility either. Then again, I guess getting a status back also doens't guarantee you can successfully perform reads or writes.

closed 2024-04-28 by mycroes

xxxxiaohuiiii, 2024-04-28:
> 我想这之前已经指出过了，但我在这里重复一遍。
> 
> 不能依赖连接状态。该属性直接返回属性。此属性仅反映最后一条 TCP 消息是否已成功传递。这种方法的缺点是 TCP 连接可能仍处于活动状态，而 PLC 实际上不会接受任何消息，但总的来说，我认为 PLC 在达到它不会执行任何进一步通信的点时会关闭连接（例如，当 TSAP 标识符不匹配时）。`Plc.IsConnected``TcpClient.IsConnected`
> 
> 如果您想知道 PLC 是否响应，只需执行您想要执行的操作（即执行您打算执行的读取）。当然，这也可能导致异常（即读取无效地址），但首先检查连接状态，然后调用要调用的任何方法将产生相同的结果。
> 
> 如果上述方法不合适（即，您的应用程序依赖于用户操作，并且您想在某个时间间隔内检查连接是否正常），您可以调用 或 ，我想只要连接正常，这些应该始终无一例外地工作。另一方面，如果它们不起作用，我也无法想象读取或写入数据的可能性。话又说回来，我想恢复状态也不能保证您可以成功执行读取或写入。`Plc.ReadStatusAsync()``Plc.ReadStatus()`

Understood, thanks for the introduction.
That's why the heartbeat signal was introduced

xxxxiaohuiiii, 2024-04-28:
> I think this has been pointed out before, but I'll repeat it here.
>
> You can't rely on connection state. The `Plc.IsConnected` property directly returns the `TcpClient.IsConnected` property. This [...]

Understood, thanks for the introduction.
That's why the heartbeat signal was introduced