S7NetPlus/s7netplus #527: 为什么结构体包含DateTime字段，使用ReadStructAsync会出现无线递归导致的应用程序崩溃？

keeleycenc, 2024-04-22:
错误信息：
System.StackOverflowException
  HResult=0x800703E9
  Message=Exception_WasThrown

尝试了很多方法，结构体还是无法有DateTime的字段
```c#
public System.DateTime ProductExpiryDate;
```

只能删除结构体中的所有DateTime的字段，才能读取成功，ReadStructAsync不支持读取数据块中的Date和Tate_And_Time以及LDT这些时间类型的数据吗？为什么文档说明没有提到？而且文档中提到：
**“read struct”和“write struct”方法不支持字符串**  

但是通过Attribute标记就可以读取啊，但是没有关于时间类型的Attribute。

```C#
[S7String(S7StringType.S7String, 50)]
public string ProductSerialNumber;
```

gfoidl, 2024-04-22:
@keeleycenc please write the issue in English, so everyone can read it (w/o the use of translators of more or less good quality), and then it's way easier to be actionable on that issue. 
Thanks!

keeleycenc, 2024-04-22:
@gfoidl Thank you for your response and guidance! I have updated my question to English as requested and added some additional details to more accurately describe my problem and use case. Please take a look at the revised question, and I hope to receive further assistance or suggestions. Thank you again for your support and patience!

```c#
var result = await _plc.ReadStructAsync(typeof(T), db, startByteAdr);
```