S7NetPlus/s7netplus #530: S7net plus with VIPA PLC

NickVerboven, 2024-06-14:
Hi

I have a .net application (written in C#) that communicates every 2 seconds with a VIPA plc.  That application is running in production evironment since 2021 and uses s7netplus release 0.7.0.  In the loggings I see that the connection to the plc sometimes is broken.  This causes problems while making a batch in production.  This issue is known since implementation, but now the problem occurs more often.    Is the s7netplus library compatible with a VIPA plc?  Is this connection problem (suddenly no connection anymore with the plc) known?
In the application 2 plc connections are used: 1 connection for logging that is opend all the time for loggings, 1 connection for starting a production batch (where information comes from MES) that is only opened when the batch is started en closed afterwards.
Any help is useful for us.
  
Thanks

MicBotta, 2025-02-12:
Hi @NickVerboven , 
I am also having communication issues with VIPA plcs, did you find out something?
In your case the communication is restored by the library (close/open) or you need to restart the PLC?

NickVerboven, 2025-02-12:
> OPGELET! Deze e-mail is verzonden van buiten AP Hogeschool. Klik geen links aan of open geen bijlagen, tenzij je de afzender kent en er zeker van bent dat de inhoud veilig is.

Hi
Restart plc is not necessary for me. Restart the .net application can help.
I did not found the root cause.
Kind regards

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>