S7NetPlus/s7netplus #532: ReadClass less than 1 byte and therefore cannot be read

sasasasa1122, 2024-07-03:
Hello, I am making a small trial program. Can you help me fix this error?
Class size must be more than 1 byte.
[image]
[image]
[image]

NeKoRoM, 2024-09-04:
i have the same issue

sasasasa1122, 2024-09-05:
İf you delete 'static' in your class, you can keep going without problem. 
Have a nice day!