S7NetPlus/s7netplus #539: What is the reason for the slow writing of variables using WriteAsynchronous?

xuxiaokang0723, 2024-08-24:
Using this function WriteAsynchronous (params DataItem [] dataItems), less than 10 variables are written, but it is very slow! What's wrong with not throwing anomalies but being slow?

xuxiaokang0723, 2024-08-24:
I use the WriteAsynchronous (params DataItem [] dataItems) function in the s7netplus library, which writes no more than 10 variables, but it is very slow. The code takes half a day to respond after executing this sentence, and the PLC also takes half a day to receive data. Why is this?

mycroes, 2024-08-25:
Just to be clear, all you're doing is opening a connection and then performing `WriteAsync`? There's no sense in that taking half a day, in all honesty there's no sense in that taking more than 10ms. However, the library applies queueing internally. If you were to queue up to half a day of reads and then perform the write, then this is very much expected (although not desired) behavior.

If you can provide a minimal reproduction of your issue I'd be happy to take a look.