S7NetPlus/s7netplus #545: .Net8 AOT feature

ypeng-ming, 2024-10-11:
Hello, 
Is it possible to compile s7netplus with .Net8 AOT feature? Cause I hear AOT feature can't be compatible with Reflection.

gfoidl, 2024-10-11:
AOT isn't officially supported by S7NetPlus at the moment.

When you don't use `class` or `struct` it could / should work with AOT, but as said not officially supported.
I mean you can try it out, and if there are troubles it would be great when these are reported (here or in another issue), so we can address these.

A similar project (from one of the maintainers here) is [Sally7](https://github.com/mycroes/Sally7). While also not officially supporting AOT it should just work, when my memory serves well.

ypeng-ming, 2024-10-12:
> AOT isn't officially supported by S7NetPlus at the moment.
>
> When you don't use `class` or `struct` it could / should work with AOT, but as said not officially supported. I mean you can try it out, [...]

Ok, Thanks for your answer, use class or struct it very power, but sometimes, WriteMultipleVariable function is alse very convenient.
I would like to try to make this AOT feature available, I'm reading and learn this project, I wish I could get it done.

closed 2024-10-12 by ypeng-ming