S7NetPlus/s7netplus #546: unity scriptler arası aktüel veri aktarımı problemi!

Boogeyman16, 2024-10-15:
unityde 3 script kullanıyorum. 

1-)Baglan (PLC' ye bağlanmak için ve veriler çağırmak için)
2-)Veriler (Baglan scriptinde çağırdığım verileri buraya yazdırıyorum ve diğer scriptlerden çağırmak için kullanıyorum)
3-) ssss (Veriler scriptinden veriyi alıp kullanmak için)

Baglan scriptinde PLC ye bağlanıp verileri "Veriler" scriptindeki "Robot2veri"class' ına yazdırıyorum. ve Baglan scriptinde "robot2veri" class' ından  çağırdığım veriyi anlık olarak okuyabiliyorum. Lakin "ssss" scriptinde "Veriler" Scriptindeki "robot2veri" class' ından çağırdığım veri '0' olarak geliyor. Değer değişmiyor.

Yardımcı olurmusunuz...
Gerekli ekleri sizlere paylaşıyorum. 
[image]
[image]
[image]

gfoidl, 2024-10-15:
@Boogeyman16 can you please type your issue in English, so that we can read it and make this issue more actionable?

Thanks in advance.

Boogeyman16, 2024-10-15:
> @Boogeyman16Lütfen sorununuzu İngilizce olarak yazabilir misiniz, böylece okuyup bu sorunu daha eyleme dönüştürülebilir hale getirebiliriz?
> 
> Şimdiden teşekkürler.

Of course 
but sorry for my english.

I use 3 scripts in Unity.

1-)Baglan script (to connect to PLC and call data)
2-)Veriler script (I write the data called in the Baglan script to this script)
3-)ssss script (To get and use the data from the Veriler script)

In the Baglan script, I connect to the PLC and write the data to the "Robot2veri" class in the "Data" script. And I can instantly read the data I call from the "robot2data" class in the Baglan script. However, the data I call from the "robot2data" class in the "Veriler" script in the "ssss" script comes as '0'. The value does not change.

Can you help me...
I am sharing the necessary attachments with you.

Thank you for your support and interest.

janbiehl, 2024-10-16:
I havent touched unity for long time. But i do think that there is no reference between the robot2veri object in 'ssss' script and in 'Baglan' script.

You should either add a public reference to your Baglan instance and access their rb2veri instance. (You commented that line out).
Or you could create a ScriptableObject (see below) to connect these two instances independently (This would be my recommendation).

```
[CreateAssetMenu]
public class Robot2Veri : ScriptableObject
{
    public float rb2_yatay_pos;
    public float rb2_dikey_pos;
}
```

In the engine you have to create a Robot2Veri instance via the "Assets/Create" menu. Now you should be able to link that instance by using the below snippet inside the required scripts (Baglan & ssss)

```
public Robot2Veri rb2veri;
```

Boogeyman16, 2024-10-17:
> I havent touched unity for long time. But i do think that there is no reference between the robot2veri object in 'ssss' script and in 'Baglan' script.
>
> You should either add a public reference to [...]

Thanks for tour help.

I wrote the code example you gave.
`[CreateAssetMenu]
public class Robot2Veri : ScriptableObject
{
    public float rb2_yatay_pos;
    public float rb2_dikey_pos;
}`

and I created a robot2veri instance in the "Assets" section. But I don't know if I should take another step?
Now Unity gives me an error and does not show the values.

I am sharing the necessary attachments with you.
[image]
[image]
[image]