S7NetPlus/s7netplus #548: When the PLC power is off, no exception is thrown

alitahaozturk, 2024-11-25:
In my project, where I use the example code shared below, I always start by opening a connection using the baglantiAc() method. Then, I use the DegerOkuInt() method to check if the connection is established (isconnected) and to read values. Generally, there is no issue, but when the power to the PLC is off and the connection cannot be established, no exception is thrown. The application freezes at that point. After restarting, it reconnects and works again. This situation always requires my manual intervention. How can I ensure an exception is thrown when the connection is lost?

[image]

ynioba, 2024-11-25:
这是来自”麦冬@广西机械工业研究院”的邮箱自动回复邮件。您好，您的邮件已收到，我查阅后尽快回复您。

lvlaimao, 2025-02-19:
When you fail to read or write, you need to write reconnect logic